daglem/reDIP-SID #3: 3v3 reg enable pin

bit-hack, 2021-06-10:
Hi there,

I was checking over the schematic and the design.  I have one small concern about the design surrounding the 1.2v regulator and the 3.3v regulator.  You have the 1.2v output driving the 3.3v enable pin.  Checking the datasheet for the 3.3v regulator however shows:

> Chip enable: Applying VEN < 0.4 V disables the regulator, Pulling VEN > 1.2 V enables the LDO

My concern is that the 1.2v output driving this pin is right on the threshold and if the 1.2v dips it could cause the 3.3v to turn off.  I seem to recall the ICE40 datasheet showing that the powerup sequence should have 1.2v starting before 3.3v which is why I guess its wired like this.

Would a simple RC circuit be better in this case driving the enable from the 5v rail?

bit-hack, 2021-06-10:
For reference this is the datasheet I was looking at: https://www.mouser.co.uk/datasheet/2/308/1/NCP167_D-2316956.pdf

daglem, 2021-06-10:
Great that you're looking into the design in such detail!

Yes, the bringup of the LDOs attempts to follow the power-up supply sequence recommended in the iCE40 UltraPlus datasheet.

I did actually read the NCP167 datasheet quite carefully, and I landed on that the design would be quite safe.
The wording in the datasheet indicates that there must be some leeway regarding the 1.2V figure: "If the EN pin voltage >1.2 V the device is *guaranteed* to be enabled." If you look at Figure 14 (Enable Voltage Threshold vs. Temperature), it would indeed seem that the 1.2V figure is extremely conservative.

closed 2021-06-10 by daglem