daglem/reDIP-SID #6: about SN74CBT16211A

sorgelig, 2022-12-02:
Hi!

On schematics and BOM you've used CBT version with 5V TTL output. Why you are insisting on this particular version instead of CBTD or CB3T versions?

daglem, 2022-12-04:
The CBT bus switch has a voltage drop of approximately 1V, hence a VCC of approximately 4.3V should yield the required 3.3V I/O.

The schematic documents how the voltage divider can be configured for CBT, CBTD, or CB3T bus switches.

closed 2022-12-04 by daglem

sorgelig, 2022-12-05:
I understand the difference between all 3 versions. Just wonder if there is a specific reason to use exactly CBT version. Or you just used what you had in inventory?

Another question: about 128mbit FLASH. I guess it's possible to use 64Mbit version, right? I have flash programmer which supports W25Q64 but not W25Q128 and i don't have Diamond Programmer.

daglem, 2022-12-05:
The reason why the board is designed to support different bus switches is to increase the probability of getting hold of all the required components during chip shortages. The CBT switch had the highest stock and the lowest price at the time, and thus the BOM contains that. As mentioned, the board should work just fine with any of the bus switches, provided that the voltage divider is configured correctly. The previous version of the design contained the CBTD bus switch, and at least one person has tested the current design with the CBT bus switch.

I don't see any reason why the 64Mbit flash wouldn't work.