daglem/reDIP-SID #7: dfu-util doesn't update

sorgelig, 2023-03-28:
Hello! 
I'm trying to use dfu-util:

```
dfu-util.exe -d 1d50:6159,:6156 -a 0 -D redip_sid.bin -R
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Warning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1d50:6159
Device DFU version 0101
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
Device does not implement get_status, assuming appIDLE
Device really in Run-Time Mode, send DFU detach request...
Device will detach and reattach...
Lost device after RESET?
```

reDIP-SID disappear from device manager and left in state when CDONE LED ON.

daglem, 2023-03-30:
It should look like this (on Linux):
```
dfu-util -d 1d50:6159,:6156 -a 0 -D redip_sid.bin -R
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1d50:6156
Device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0101
Device returned transfer size 4096
Copying data from PC to DFU device
Download	[=========================] 100%       104090 bytes
Download done.
DFU state(2) = dfuIDLE, status(0) = No error condition is present
Done!
Resetting USB to switch back to Run-Time mode
```
It seems like you don't get the device status from the bootloader, for some reason. I guess this could be caused either by dfu-util on your OS or the OS itself (I can't help with Windows), or by an incorrect bootloader. The bootloader should be built like this:
```
cd no2bootloader/gateware/ice40-stub/
make BOARD=redip-sid PRE_CLEAN=1 bootloader-clean
make BOARD=redip-sid bootloader
```
If you're using an FTDI cable / board, you would use iceprog for programming flash:
```
sudo iceprog build-tmp/bootloader.bin
```

sorgelig, 2023-03-31:
i use muacm.v netlist file from redip-sid repo. I guess it's pre-compilled netlist by you. Is it correct?
Currently i use my ROM Programmer G540 over that 10-pin connector. It's very very slow, but works. I hope i can make dfu working.
What driver version you use?

sorgelig, 2023-03-31:
wait... i didn't get it.. I flash redip-sid.bin into EEPROM from address 0. Do i have to burn bootloader.bin? From which address?

sorgelig, 2023-03-31:
```
make BOARD=redip-sid PRE_CLEAN=1 bootloader-clean

../build/project-rules.mk:56: c:/Work/oss-cad-suite/no2bootloader/gateware/ice40-stub/build-tmp/proj-deps.mk: No such file or directory
process_begin: CreateProcess(NULL, env bash C:\Work\oss-cad-suite\bin\yosys-config --datdir/ice40/cells_sim.v, ...) failed.
makefile:52: *** multiple target patterns.  Stop.
```

i've checked, there is really no no2bootloader/gateware/ice40-stub/build-tmp/proj-deps.mk
Even in repository there is no such file.
I've tried to make an empty one, but still no go.. 

```
make BOARD=redip-sid PRE_CLEAN=1 bootloader-clean
process_begin: CreateProcess(NULL, env bash C:\Work\oss-cad-suite\bin\yosys-config --datdir/ice40/cells_sim.v, ...) failed.
makefile:52: *** multiple target patterns.  Stop.
```

sorgelig, 2023-03-31:
by the way, may be it worth to connect ST pin from LM66200DRLR to I2C_SDA_BTN? it will automatically press the button and enter into dfu mode when USB is being connected.

sorgelig, 2023-04-01:
May be keep this issue opened to get more visibility for those who will also struggle with correct ROM flash?
Or update readme with short instruction and then it can be closed.

daglem, 2023-04-03:
I'll add instructions for flashing a blank board later, for the few people who assemble their own boards.

Regarding your point 3 above: You should be able to use muacm if you like, then you just use `dfu-util` without having to do the press button / powerup dance first. However since I still haven't been able to fit two SIDs + muacm on the FPGA, I have given instructions without muacm.

daglem, 2023-04-05:
Heh, you're right, something is off now. muacm used to work just fine. I'll investigate what could have changed, as I don't see off-hand what that might be.

muacm is a separate project. The source code for muacm can be found at https://github.com/no2fpga/no2muacm, however I have used a slightly modified pre-built version from https://github.com/no2fpga/no2muacm-bin

daglem, 2023-04-05:
> by the way, may be it worth to connect ST pin from LM66200DRLR to I2C_SDA_BTN? it will automatically press the button and enter into dfu mode when USB is being connected.

The USB interface is intended to be generic, e.g. it can conceivably be used for things like USB audio. Because of this I won't make changes to specifically tailor it to flashing.

muacm is intended for simplicity of flash upgrades (just connect and flash, regardless of whether the board was powered up first). I just have to get it working again.

daglem, 2023-04-05:
> MUACM doesn't work, actually.. So my original issue is exactly for this case. Once i've compiled and flashed bootloader it started to work with dfu-util. May be muacm precompiled netlist has compatibility issue. Do you have instruction how to produce muacm.v netlist from no2bootloader? Then i will make it from current bootloader sources.

muacm should work again now. Note that you still need the bootloader installed, muacm boots into that in order to flash.

closed 2023-04-05 by daglem

sorgelig, 2023-04-05:
It doesn't fit into FPGA now. Strange, it was possible to compile before.
Anyway, it's not a big problem to press the button :)

daglem, 2023-04-05:
Currently, muacm will only fit if you choose the same model for both SIDs (i.e. both MOS6581 or both MOS8580).

I'm working on it :sweat_smile:

sorgelig, 2023-04-05:
Ah, that explains why.. I've used muacm before i started to play with 8580/6581 external switch. I think having muacm is not as important as having bug-less SID emulation.
It seems not all auto-detections of SID type work in reDIP-SID. I've tried Elvis wave playback cartridge - it detects reDIP-SID as 6581 always regardless type of emulated SID.

daglem, 2023-04-05:
Never heard of that cartridge - do you happen to have the SID model detection code it uses available?
Feel free to create separate issues for any incompatibilites.

ATM the emulation should be at least on par with reSID 0.16, with good 8580 emulation, and passable 6581 emulation. I'm working on freeing up resources which could be used for improved emulation and runtime configuration, if possible with muacm included at the same time.

sorgelig, 2023-04-05:
False alarm. Probably it was some hiccup while testing switch. I've just re-tested it - detected correct SID models.

sorgelig, 2023-04-05:
By the way, back to idea of using ST signal from LM66200. May be it can be used for quick detection of USB, so instead of muacm requiring many resources, you can check this signal and boot into bootloader. It can be made as an option.
Of course it will require a new PCB revision.. I don't know if USB connection can be detected by DP/DM signals without muacm.

sorgelig, 2023-04-05:
I just discovered that `btn` signal (from `sgtl5000_init`) can be connected to `boot` of `ice40_init`. So it's possible switch to bootloader any time which is convenient when you experiment with firmware, so you don't need to re-plug USB every time and keep connector more healthy. It's also easier than trying to hold pressed such tiny button while connecting USB.
So, may be you will consider this function as well.

sorgelig, 2023-04-05:
sorry for spamming, but i found the method not requiring pcb modifications:

1) add pullup for usb_d_p and usb_d_n
2) use ```boot <= boot | bootloader | ~btn | ~(usb_d_p & usb_d_n);```

That's all. Once you connect USB, it will immediately switch to bootloader. no muacm required.

sorgelig, 2023-04-06:
that's almost exactly what i did ;)

daglem, 2023-04-06:
Assuming that an unconnected EIR will be biased at VAG (currently configured to 1.5V), you could possibly just connect a three way switch between GND and 3.3V, with the middle pin connected to EIR. Both GND and 3.3V are available on the programming header. But perhaps adding external components is preferable to having to solder on the board? :man_shrugging:

In any case I think you would have to disable the ADC analog zero cross detector in by changing the value written to `CHIP_ANA_CTRL` in `sgtl5000_init_data.hex` (change `00 24 01 37` to `00 24 01 35`), and of course also split out the right channel input for detection instead of mixing in `redip_sid.sv`. Be warned that I've never tested whether my I2S implementation works for input :sweat_smile:

sorgelig, 2023-04-06:
I still prefer A5+A8 :) Stereo/dual SID is not interesting to me, so these pins aren't used anyway.