daglem/reDIP-SID #8: Suggestion for mono dual-chip switch

sorgelig, 2023-04-01:
Hi!
I suggest to use A5 and A8 inputs as 8580/6581 modes switch when compiled as a single SID (SID2 not defined).

i propose to changes into lines: https://github.com/daglem/reDIP-SID/blob/c4514e9e33d27e8cd482e8126472a7c23486bb5e/gateware/sid_api.sv#L74-L75

change it to:
```systemverilog
    sid::cfg_t  sid1_cfg = {  cs.a5 ? sid::MOS8580 : sid::MOS6581, sid::D400, 9'd250, -11'sd384 };
    sid::cfg_t  sid2_cfg = { ~cs.a8 ? sid::MOS8580 : sid::MOS6581, sid::D400, 9'd250, -11'sd384 };
```
{A8,A5} will do:
11 - 8581+6580
01 - 2x6581
10 - 2x8580

So, if user will connect 3-positions switch to A8 and A5 with middle pin to GND, then in middle position it will be a mix of 6581 and 8580 and on side positions it will be either 6581 or 8580. Similar to SIDFX board.
in pcf file these pins need to be pulled up which shouldn't harm as pullup resistor is very weak (20K-100K). So it won't require external pullup resistors.

I believe 99% users will need this mode. Generic user just need to play games and thus simply need to switch between 6581 and 8580 in real time. 

I've also added -384 fc_offset, so 6581 mode sounds more generic way like 6581R2/R3 (R4, R4AR are muffling IMHO).

If you want, i can make pull request with required changes.

daglem, 2023-04-03:
The plan is to add software configuration (ideally with two SIDs), however I'll consider your switch idea for single SID gateware later, when I have finished up all the other stuff I have planned. I won't be merging any PRs for now, since that would only detract from the work I'm doing.

Since you're playing with this now, I'll just mention that the default pullup value is 100K (it shouldn't be necessary to change the pcf file), however you have to actually configure the pullup for the IO pad - see how it's done for `pad_phi2` in `sid_io.sv`.

sorgelig, 2023-04-03:
I've already checked. There is no pullup by default. If i touch pin by GND and remove then it will keep logic 0 for some time. So there is no pullup if not explicitly configured in pcf. After adding pullup in PCF, it started to work correctly.

> The plan is to add software configuration (ideally with two SIDs), however I'll consider your switch idea for single SID gateware later

Static software configuration is not convenient in mono mode. I often like to switch between 8580 and 6581 while listening the music.

daglem, 2023-04-05:
> I've already checked. There is no pullup by default. If i touch pin by GND and remove then it will keep logic 0 for some time. So there is no pullup if not explicitly configured in pcf. After adding pullup in PCF, it started to work correctly.
> 

Could it be that you didn't set the `PULLUP` as in the example I pointed to above? My point was just that the pullup value defaults to 100K _when turned on_, and that configuring on/off can be done (dynamically, if so desired) without changing the pcf file. I guess it doesn't matter much in this case.

> Static software configuration is not convenient in mono mode. I often like to switch between 8580 and 6581 while listening the music.

The plan is to add dynamic configuration via register writes, but in any case having an option for a physical switch is interesting, and I'll consider it later. In the mean time, just configure as fits you best.

sorgelig, 2023-04-05:
> Could it be that you didn't set the `PULLUP` as in the example I pointed to above? My point was just that the pullup value defaults to 100K _when turned on_, and that configuring on/off can be done [...]

I see now. I didn't understand your answer before. I will try your suggestion.

sorgelig, 2023-04-05:
Yes, `.PULLUP(1'b1)` for `io_cs` works fine too.

vanfanel, 2024-06-13:
@sorgelig Did you consider to bring this SID engine to the MiSTer C64 core?
If not, why? Is current C64 SID more accurate?

sorgelig, 2024-06-13:
Current SID implementation in MiSTer is very accurate. It also has many useful functions such as filter customization allowing to simulate different SID revisions. It's quite mature and tested in many games.

vanfanel, 2024-06-13:
> Current SID implementation in MiSTer is very accurate. It also has many useful functions such as filter customization allowing to simulate different SID revisions. It's quite mature and tested in many games.

That's my understanding, too. It sounds perfect to me.
So you also prefer the current SID implementation on MiSTer to current reDIP-SID... This confirms my impressions from hearing both.